tom-r/tactics_pi #1: Error during compilation

ptulp, 2016-06-27:
Thomas,
During the compilation for OSX I get the following error.
/Users/petertulp/Devel/Plugins/tactics_pi/src/tactics_pi.h:263:38: error: extra qualification on member 'ApplyPrefs'
      void TacticsPreferencesDialog::ApplyPrefs(wxCommandEvent& event);

In tactics_pi.cpp line 3246 and line 3257 you call ApplyPrefs without parameter.

Peter

tom-r, 2016-06-27:
Peter,

I guess it's tactics_pi.h, line 263

The call is always done without parameter in these cases, ex. L. 2871, tactics_pi.h
This is "standard" dashboard_pi ...
Thomas

ptulp, 2016-06-27:
Thomas,

When I left out the TacticsPreferencesDialog:: part from line 263, the compilation went OK and the plugin build for OSX.

Peter

tom-r, 2016-06-27:
OK,

I'll adopt the code and commit again with the change in approx. half an hour.

Thomas

tom-r, 2016-06-27:
Peter,

I adopted the class definiiton as discussed above.

Thomas

closed 2016-06-27 by tom-r