tom-r/tactics_pi #4: Project description missing

SamsonovAnton, 2016-07-08:
It would be nice to cover the purpose of this plugin in `README.md` and also provide a short description of the repository.

tom-r, 2016-07-24:
Anton,

next to README.md you find the detailed documentation in tactics_pi.odt and tactics_pi,pdf.
I mentioned both in README.md now.

Thomas

closed 2016-07-24 by tom-r

SamsonovAnton, 2016-07-25:
> You find the detailed documentation in tactics_pi.odt and tactics_pi,pdf. I mentioned both in README.md now.

That is all fine, but I meant that the entire introductory part of `tactics_pi.pdf` should be copied to `README.md` prologue, — so that people know _**why**_ should they be interested in this plugin in the first place. A brief summary would also be great as a project description here on GitHub, instead that default “No description or website provided” message.

tom-r, 2016-08-10:
Anton,

I updated README.md and poject description

Thomas