tom-r/tactics_pi #6: The SVG icon is missin in Journeyman tools style

Hakansv, 2016-07-13:
The icon is nice in traditional style but has to be adapted also to Journeyman. See attached pictures:
Unfortunately I've no advice how to do this but most of the other plugins are now working.

[image]
[image]

tom-r, 2016-07-14:
Hakan,
I'm underway sailing till 24.th of July. I will come back to you after my return
Thomas

tom-r, 2016-07-24:
Hakan,
I added the standard icon to the github code for Journeyman styles.
Thomas

closed 2016-07-24 by tom-r

Hakansv, 2016-07-25:
Thomas
Thanks, as nice as the trad. one.
BTW Please give a note when you're ready for latest language updates. Some more are performed since last push, but French is not completed. I think his out sailing. (As me for the very moment.) Use: momo at stupan.se
Håkan