tom-r/tactics_pi #21: Negative True Headings.

trudK45, 2018-03-19:
When sailing close to north, say 350 degrees and you have a variation of for example  West 15 degrees you get negative true headings. Input is HDG and WMM is used to establish variation.
This error exist when having HDG or HDM as compass input.
I have modified the code and tested this and it now works as it should.
I can make a branch and commit if you want.

tom-r, 2018-03-20:
Thanks for fixing, please give me time till the next weekend, I don't have access to my development machine right now.
BR,
Thomas

trudK45, 2018-03-20:
If you give me push access I can make a remote branch and do a push. Only 2 files are changed tactics_pi.cpp and tactics_pi.h. You will have to change the revision I suppose.
/Hans

trudK45, 2018-03-22:
I did a fork instead and made a pull request. #22

rgleason, 2018-03-22:
trudik45, I would like to personally thank you for your persistence in getting this fixed for Opencpn, Dashboard and Tactics.  I realize that you are responding to several reports by users. Thank you for figuring out what was happening and providing the programming!

rgleason, 2018-03-24:
Thanks very much Thomas, I'll  compile, test and put it up for others, with the new language files and trudK improvement.

tom-r, 2018-03-24:
Rick.
I'll do a new patch with this change. I just had some issues compiling yesterday evening and couldn't finish.
Just wait until this is done (probably tonight).
BR,
Thomas

rgleason, 2018-03-24:
I think we want to find these binaries  OpenCPN_buildwin-up_to_4.8.0.7z    27-12-2017   7mb.
 at this location https://sourceforge.net/projects/opencpnplugins/files/?source=navbar

I haven't used this package, I have the binaries that were provided earlier.
Perhaps I should update?   I don't believe they've changed but perhaps they have?
Pavel is a man of few words, he doesn't tell us much.

rgleason, 2018-03-24:
I am going to update to  Correction: OpenCPN_buildwin.7z | 2017-12-27 | 6.2 MB

rgleason, 2018-03-24:
Unless you are having a problem.

rgleason, 2018-03-31:
Thomas, you did it!  https://github.com/tom-r/tactics_pi/releases
Was it that "release" directory.

rgleason, 2018-05-04:
Thomas, TransmitterDan has some changes to trudK45 improvements in Opencpn. They may apply too Tactics too.

See https://github.com/rgleason/OpenCPN/commit/c4fb6a0ad0205501ae902b57f9c64b7d0262a199

rgleason, 2019-07-08:
I think this can now be closed. .. Is that correct?

tom-r, 2019-07-08:
Hi Rick,
yes it can be closed.
I still had trudK45's code in tactics, and transmitterdans code is exactly the same functionality, just a bit more "elegant" ...
So I changed it manually to transmitterdans code (local "heading" variable).
It'll come with the next patch, I'm reworking some code on the same file right now, which is not yet ready for a push to github ...
Thomas

closed 2019-07-08 by tom-r

rgleason, 2019-07-08:
Thanks so much. Thomas.