tom-r/tactics_pi #23: Correct cmake install path to align with Opencpn

rgleason, 2018-03-31:
Capt Ron Reports several times:
http://www.cruisersforum.com/forums/f134/tactics-plugin-166909.html#post2607345

> ..move the tactics plugin files because the location reference is different from OpenCPN.
> 
>     OpenCPN references /usr/,
>     plugin references /usr/local. 
> 
> ...I forgot and did not change the reference in the CmakeCache.txt file in the build directory where it states
> 
>     CMAKE_Install_Prefix:Path = /usr/local
> 
> if I had changed this to just "/usr"
>the built files would have ended up the correct location for OpenCPN. 

Thomas can this be fixed?

rgleason, 2018-04-01:
Capt Ron wrote:

> I see you already added an issue to Tom-r issue page, that is fine. The CmakeCache.txt file is created during the make process and is in the build directory. I don't know if the localization dependency comes from something in Tom-r's code or from the system. In the thread on 'compiling for Debian' Janth indicates that you could just change the localization after running cmake and the CMakeCache.txt file is created and before your run the plugin install commnad. I found it pretty easy just to move the files afterwards. There are two sets of files to move, the tactics_pi.so file and the tactics_pi/data files that also end up in a different directory.

rgleason, 2018-04-01:
I believe the Cmake directory has a file that can be changed.

dominig, 2018-12-11:
Current code is correct.
cmake should get the default config from the distro.
If you want anything different, then just pass the param to cmake.

rgleason, 2022-03-25:
This can be closed.

closed 2022-03-25 by rgleason