tom-r/tactics_pi #24: Feature request: store wind data and display/graph longer term wind data

LeaseOnLife, 2018-05-23:
Example use case: I would like to see diurnal wind shifts. The graphing of TWS and TWD in tactics is appealing, including the filtering. 

It would be great to be able to plot the last 24h, 48h, 72h for example, and also have past data available after opencpn has been restarted.

rgleason, 2018-05-23:
Have you considered Plots_pi?   What is wrong with that?

LeaseOnLife, 2018-05-23:
sure I did, see the plots_pi thread you created on CF and the discussion with sean I had. it does not calculate twd or twa or tws. it relies on outside systems to do it, can be nmea-converter, can be something outside of O.

tactics_pi works well for me doing the calculations, and I like the graphs for twd and tws much better, it has  filtering/smoothing while still showing the peaks, nice, very usable, intuitive. 

plots_pi looks far more simplistic, but does store data and you can plot longer term.