tom-r/tactics_pi #27: Feature Quick on/off from icons bar

dominig, 2018-12-12:
A very useful facility in AIS, is to be able to deactivate AIS by a simple click on its ICON.
It would be nice for Tactic to do so.
Some time it's important to remove all the extra from the screen to concentrate attention on important details such as rocks.
Currently is requires to make several right/left click action.
My ideal would be a toogle.
 - instruments+extra overlay
 - overlay only
 - nothing

rgleason, 2019-07-17:
Good idea.

rgleason, 2022-03-25:
Actually what happens is Toggling the Toolbar Icon turns the instruments display off and on. To remove the overlay you must uncheck  "Enable" in Toolbar > Plugins > Tactics  (uncheck enable).

This is done. Can be closed  @dominig  @tom-r