tom-r/tactics_pi #33: STW w/ Leeway correction clarification

canne, 2019-07-14:
Hi Thomas,

Maybe something to add in the documentation.

I noticed that since sometimes ([not in 2018/03](http://davidburchnavigation.blogspot.com/2018/04/effect-of-leeway-on-knotmeter-speed.html)) once one corrects the STW with Leeway, also the Leeway gets corrected (luckily not in a recursive manner! :smile: ). A bit confusing to use corrected value to correct the value which was used for the correction - when one should stop? Fine with me but perhaps good to explain this.

Another subject of confusion without a mention in documentation is that tactics uses always the corrected STW in current calculations no matter what is the selection in preferences. If one verifies current direction using the displayed, non-corrected values this may lead to a confusion. Perhaps it would be better either to have a separate control for this (_in my boat I have both ultrasonic and paddle water speed sensors and maybe I would like to install the fancy one back one day_) or to use the STW correction tick box for everything and document that.

Best regards,

Petri

tom-r, 2019-07-15:
Petri,
"I noticed that since sometimes (not in 2018/03) once one corrects the STW with Leeway, also the Leeway gets corrected (luckily not in a recursive manner! smile ). A bit confusing to use corrected value to correct the value which was used for the correction - when one should stop?"

That's not a feature, that's a bug !
The second parameter in CalculateLeeway (value) is explicitly meant for STW, but it's never used in the routine. I corrected this to use "value" instead of mStW for the Leeway calc.
And the input into CalculateLeeway is now the uncorrected original value from SendSentences to all Instruments...
I did it like this :
void tactics_pi::SendSentenceToAllInstruments(int st, double value, wxString unit)
{
  **double org_value=value;**
	if (st == OCPN_DBP_STC_AWS){
....
	SetCalcVariables(st, value, unit);
	CalculateTrueWind(st, value, unit);
	CalculateLeeway(st, **org_value**, unit);
...

canne, 2019-07-15:
Thanks, Thomas,

I remain a bit confused when you say that "_I did it like this .... org_value = value;..._" since I cannot find the "org_value" from https://raw.githubusercontent.com/tom-r/tactics_pi/master/src/tactics_pi.cpp . But if you mean that you got it working and plan to commit later, that's fine, I'll check and fetch/merge upstream/master and retest once you're happy with it. Please confirm, I'm right now writing a test plan.

The second part (forced STW correction in current calculations) - a document update is largely sufficient for now, I would say.

tom-r, 2019-07-15:
Hi Petri,
you're right, it's not yet committed.
1st : I have to test it, before I commit and hand it out to public. 
2nd : I have some other pending changes coming which will affect tactics_pi (data export for WindHistory, BaroHistory and PolarPerformance).

"forced STW correction in current calculations" : if your instruments already do the correction by themselves (on NKE or B&G it could very well be ...) then we have a double correction ...
I'll be out sailing next weekend, let's see what the current calculation looks like w/o the correction...
Thomas

canne, 2019-07-15:
Alright, I'll make a quick fix then to finish my test and then sync back with you for the same functionality from upstream.

Data export: that's interesting, finally some time domain! Have you considered InfluxDB + Grafana?

tom-r, 2019-07-16:
Data export:
No not yet, right now it's gonna be a simple text export, with a selectable exportrate (in steps) & separator (via ini) 
Thomas

canne, 2019-07-16:
_Data export_: In view of data import into a database -  may I suggest that you change _Now()_ to [UNow()](https://docs.wxwidgets.org/trunk/classwx_date_time.html#a0db1c9bac7b2d27d589cad2a38d1438c) in
` m_ArrayRecTime[WIND_RECORD_COUNT - 1] = wxDateTime::Now().GetTm( );`
and that in exported CSV file that timestamp would be exported from a uint64 in milliseconds from EPOCH ([Unix time](https://en.wikipedia.org/wiki/Unix_time)) - for clarity, UTC format is OK.

_Current calculation STW correction_: My ultrasonic Poncin (the distributor) is fully compatible with Airmar paddlewheel - Raymarin ST60 does not know the difference and it makes no correction. But for ultrasonic, I do not need the correction, anyway. I would prefer that if the "correct STW" box is not ticked, no correction in STW whatsoever will take take place.