tom-r/tactics_pi #34: Resolving changes - Canne-master to Tom-r-master

rgleason, 2019-07-28:
It looks like some of Petri's initialization changes May 11 and maybe some appveyor changes got lost?
https://github.com/tom-r/tactics_pi/compare/master...canne:master

Going in the other direction there seem to be many more differences...
https://github.com/canne/tactics_pi/compare/master...tom-r:master

There are conflicts so I don't feel comfortable reviewing it.

tom-r, 2019-07-28:
Rick,
I created patch 10 in tactics_pi which includes the Export functionality for BaroHistory, WindHistory & PolarPerfomance. 
BaroHistory was not done by me, but contributed by another user. I contained a lot of dead code (unused variables, exponential smoothing of the pressure data without using it at all, etc). It looked like a copy of the wind history instrument, and the code was not converted completely, but a lot of stuff was left unused from wind history ...
Adding the export functions, I cleaned BaroHistory up (at least the stuff I noticed directly).
The 'Export changes' are not yet included in Petris plugin. We discussed this yesterday.
He's in the middle of testing and will merge afterwards.
Thomas

canne, 2019-07-28:
Hi Rick,
I think you can forget those commits from May: In June, I forked _dashboard_pi_ and stopped working on my _tactics_pi_ repository. I am not modifying it and I will not make any pull requests to _tactics_pi_  until further notice, I do not have the time necessary for that. If I'll find a bug, I'll report it to Thomas, of course, but he cannot sync with me, since I have _dashboard_pi_ derivation and he has got _tactics_pi_...
Same the other away around: I can pull in his changes into Tactics classes only manually. Because of time constraints, I'll do that only when Thomas publish a new _tactics_pi_ release with some new features.
Let's hope we have time for that. I have winded my development down and I am planning to stop it with a fully tested release version in a month or so. After which I move to the other side of _**O**_: I want to improve the sloppy data acquisition :grin: , storage and post-processing, cf. history saving features Thomas mentioned.
Best regards,
Petri

rgleason, 2019-07-28:
Petri, I understand.  So, we will just be building tactics and dashboard_tactics separately. So closing.

closed 2019-07-28 by rgleason