tom-r/tactics_pi #38: PI -Shouldn't user see version major.minor.patch?

rgleason, 2019-07-30:
Under preferences we just see Mayor.Minor

canne, 2019-07-30:
This is the same thing for Dashboard. What says the OpenCPN developer's guide?

canne, 2019-08-02:
OK, I fixed it, but in Dashboard only and it is not perfect, since there is only
```
      virtual int GetPlugInVersionMajor();
      virtual int GetPlugInVersionMinor();
```
in API (ocpn._plugin.h - see https://raw.githubusercontent.com/OpenCPN/OpenCPN/master/include/ocpn_plugin.h )
So we need a workaround. This is what I did. Dashboard's VERSION strings are hard-coded and the _version.h_ file is useless and total fake. So please see below `#ifdef _TACTICSPI_H_` to see how I fixed the entire thing to be consistent with the build package name in Dashboard modules.

Let's start with the VERSION.cmake. It looks now like this:
```
SET(VERSION_MAJOR 0)
SET(VERSION_MINOR 4)
SET(VERSION_PATCH 001)
SET(VERSION_DATE "2019-07-31")
##############################
SET(PLUGIN_NAME_SUFFIX "ov50")
SET(PLUGIN_VERSION_MAJOR "${VERSION_MAJOR}")
SET(PLUGIN_VERSION_MINOR "${VERSION_MINOR}")
SET(PLUGIN_VERSION_PATCH "${VERSION_PATCH}")
SET(PLUGIN_VERSION_DATE ${VERSION_DATE})
```
The upper part is for the compilation (for the C-code) while the lower part repeats it for the Cmake build package. Talking about cmake:
_cmake/version.h.in_ looks like this now:
```
#define PLUGIN_VERSION_MAJOR ${VERSION_MAJOR}
#define PLUGIN_VERSION_MINOR ${VERSION_MINOR}
#define PLUGIN_VERSION_PATCH ${VERSION_PATCH}
#define PLUGIN_VERSION_DATE ${VERSION_DATE}
```
Oh, there is a _version.h_ in the _src_-directory. Did I mention it is useless? We're going to use one generated from the _cmake/version.h.in_. **Delete _src/version.h_**!

But 🤦‍♂ still does not use _version.h_...No wonder, since both Dashboard and Tactics are not using it but there is hard-coded version number in _dashboard_pi.h_ (and in _tactics_pi.h_). Let's modify that a bit:
```
#ifdef _TACTICSPI_H_
#include "version.h"
#else
#define     PLUGIN_VERSION_MAJOR    1
#define     PLUGIN_VERSION_MINOR    5
#endif // _TACTICSPI_H_
```
If you are doing for tactics_pi, then in _tactics_pi.h_ you would replace
```
#define     PLUGIN_VERSION_MAJOR    1
#define     PLUGIN_VERSION_MINOR    0
#define     PLUGIN_VERSION_PATCH    10
```
with
`#include "version.h"`
From where that _version.h_ is coming from, BTW? It will be generated in _build/CMakeFiles/include_ .

Fair enough, what about **_O_**, who does not know about patch levels it plugin API? This is my suggestion:
In _dashboard_pi.cpp_ (in my case, in your case _tactics_pi.cpp_):
```
int dashboard_pi::GetPlugInVersionMajor()
{
    return PLUGIN_VERSION_MAJOR;
}

int dashboard_pi::GetPlugInVersionMinor()
{
#ifdef _TACTICSPI_H_
    int version_patch = (PLUGIN_VERSION_MINOR * 1000) + PLUGIN_VERSION_PATCH;
    return version_patch;
#else
    return PLUGIN_VERSION_MINOR;
#endif // _TACTICSPI_H_
}
```
Of course, in _tactics_pi.cpp_ it is enough to say:
```
int dashboard_pi::GetPlugInVersionMinor()
{
    int version_patch = (PLUGIN_VERSION_MINOR * 1000) + PLUGIN_VERSION_PATCH;
    return version_patch;
}
```
Which will give as a result, for dashboard_tactics_pi v0.4.001 : 
[image]
This is closest I can get to show user major.minor.patch. Otherwise, ask the **_O_** gods to give us, sheer mortals a _GetPluginVersionPatch()_ API call 😉 !

rgleason, 2019-08-03:
Petri, Thank you for your code suggestions and analysis. I understand now why Bdbcat wrote what he did., an api change is needed and the plugin needs to change. Also, thanks for the workaround, I think i will be good enough for until the API is changed.

I will delete version.h, I think that is left over from the way it was done before.

rgleason, 2019-08-03:
VERSION.cmake
SET(PLUGIN_VERSION_MAJOR "1")
SET(PLUGIN_VERSION_MINOR "0")
SET(PLUGIN_VERSION_PATCH "010")
SET(PLUGIN_VERSION_DATE "2019-07-26")
SET(PLUGIN_NAME_SUFFIX "ov50")
##############################
SET(PLUGIN_VERSION_MAJOR "${VERSION_MAJOR}")
SET(PLUGIN_VERSION_MINOR "${VERSION_MINOR}")
SET(PLUGIN_VERSION_PATCH "${VERSION_PATCH}")
SET(PLUGIN_VERSION_DATE ${VERSION_DATE})
------
version.h.in
#define PLUGIN_VERSION_MAJOR ${VERSION_MAJOR}
#define PLUGIN_VERSION_MINOR ${VERSION_MINOR}
#define PLUGIN_VERSION_PATCH ${VERSION_PATCH}
#define PLUGIN_NAME_SUFFIX "${NAME_SUFFIX}"
#define PLUGIN_VERSION_DATE "${VERSION_DATE}"
----------------
tactics_pi.h
//#define     PLUGIN_VERSION_MAJOR    1
//#define     PLUGIN_VERSION_MINOR    0
//#define     PLUGIN_VERSION_PATCH    10
//Note: Version.h will be generated in build/CMakeFiles/include.
#include "version.h"
----------
tactics_pi.cpp
//int tactics_pi::GetPlugInVersionMinor()
{
//    return PLUGIN_VERSION_MINOR;
}
int tactics_pi::GetPlugInVersionMinor()
{
    int version_patch = (PLUGIN_VERSION_MINOR * 1000) + PLUGIN_VERSION_PATCH;
    return version_patch;
}
--------------
Thanks. We'll see what travis and appveyor produce.

canne, 2019-08-03:
`#define PLUGIN_VERSION_DATE "${VERSION_DATE}"`
in version.h.in should be
`#define PLUGIN_VERSION_DATE ${VERSION_DATE}`
because in VERSION.cmake you already have a string definition
`SET(PLUGIN_VERSION_DATE "2019-07-26")`

rgleason, 2019-08-03:
Well, it failed last night...as predicted.  I did wonder about those quotes.  I am still trying to figure out what acts on what with C++
 ```
tactics_pi.cpp
C:\project\tactics_pi\src\tactics_pi.cpp(747): error C2561: 'tactics_pi::GetPlugInVersionMajor': function must return a value [C:\project\tactics_pi\build\tactics_pi.vcxproj]
  C:\project\tactics_pi\src\tactics_pi.cpp(745): note: see declaration of 'tactics_pi::GetPlugInVersionMajor'
C:\project\tactics_pi\src\tactics_pi.cpp(751): error C2447: '{': missing function header (old-style formal list?) 
```

rgleason, 2019-08-03:
VERSION.cmake
Upper part for C precomiler, uses integers except for date & name_suffix,  injected into version.h.in 
Lower part for CMAKE date & name suffix not strings and major,minor,patch are strings. 

Regarding VERSION.cmake lower part, we never had that before. This is done elsewhere for CMAKE, why are we doing it here?  --So we can have these values available for the plugin's Preferences?

Well, that did not build either, perhaps I have something wrong?
https://ci.appveyor.com/project/rgleason/tactics-pi/builds/26446715
```
CMake Error at VERSION.cmake:11:
  Parse error.  Function missing ending ")".  Instead found unterminated
  string with text ")
  SET(PLUGIN_VERSION_DATE ${VERSION_DATE})
  SET(PLUGIN_NAME_SUFFIX ${NAME_SUFFIX})".
Call Stack (most recent call first):
  cmake/PluginConfigure.cmake:87 (include)
  CMakeLists.txt:32 (INCLUDE)
-- Configuring incomplete, errors

```

canne, 2019-08-03:
> Upper part for C precomiler, uses integers except for date & name_suffix, injected into version.h.in

Correct
> Lower part for CMAKE date & name suffix not strings and major,minor,patch are strings.

Incorrect. Lower part is for CMake to feed the build package module and others in cmake directory. In this process, there are no integers, only string variables so the second part of the statement is not necessary.
> Regarding VERSION.cmake lower part, we never had that before.

Incorrect statement. We have added the upper part for the cmake/version.h.in . The lower part has always been there. Only that one does not need to type the same values again since they are defined in the upper part. This is less error prone.
> This is done elsewhere for CMAKE, why are we doing it here?

This statement I cannot understand. CMake is a program which reads in Makefiles. In which makefile in the plugin project  is the "elsewhere" where these variables are defined?
> So we can have these values available for the plugin's Preferences?

If you mean the upper part variables, yes, one can define as many variables one likes in the upper part and its counterpart macro in version.h.in. The C/C++ program which is building the Preferences dialog can use the macro-definitions, of course. But since they are static, where's  the point? I cannot see why you would introduce static, build-moment settings to the end user in Plugin's Preferences dialog? Maybe the version number? That can be helpful.

But as I said, what is lacking it the OpenCPN Plugin Programmer's Manual Style Guide which imposes to all developers the same look and feel.
> perhaps I have something wrong?

Certainly. If we look at the entire error message, it is quite self-explanatory:
```
CMake Error at VERSION.cmake:11:
  Parse error.  Function missing ending ")".  Instead found unterminated
  string with text ")
  SET(PLUGIN_VERSION_DATE ${VERSION_DATE})
  SET(PLUGIN_NAME_SUFFIX ${NAME_SUFFIX})".
Call Stack (most recent call first):
  cmake/PluginConfigure.cmake:87 (include)
  CMakeLists.txt:32 (INCLUDE)
-- Configuring incomplete, errors occurred!
See also "C:/project/tactics_pi/build/CMakeFiles/CMakeOutput.log".
Command exited with code 1
```
Makefile is parsed line by line, so the error is in the line
`SET(PLUGIN_NAME_SUFFIX ${NAME_SUFFIX})`
Since it is correct statement by it syntax, the error must be above, in `${NAME_SUFFIX}`.
Indeed, looks like that you have not closed the quotes:
https://github.com/rgleason/tactics_pi/commit/62209f879d456cb3b2c78ad91569f5fc3d924f28#r34558423
Even if it would have passed the CMake processing, the C-precompiler would have complained about it.

So, it looks to me that you are not too far, one more step. 👍 

Rick, whatever names you want to use or which ever model you want to use, it is fine with me as long as there is no mention about the correct procedure in the Plugin Programmer's Guide, or if such  document does not exist. `OCPN_MIN_VERSION` is fine, of course but does it, in the package name,  _guarantee_ that the plugin would work in 5.1 ? It is useful with this name as a macro in version.h.in only: this way the plugin can check at the startup that the hosting O is the right version and can complain to the user if no. But for that, the plugin code must have this code. If there is no such a check, the suggested name becomes questionable.

Anyway, myself I did not even had that line, I took the PLUGIN_NAME_SUFFIX from **you**! ... :wink:

Anyway, you asked for it,I delivered. If you don't like it, nothing prevents you to swap back to original **_O_** "standard" copy-paste method to get major.minor in the **_O_** plug-in dialog.

rgleason, 2019-08-03:
This is good Petri.  It compiled and I tried it. Under Options>Plugins  Tactics 1.0 is what shows.
I think I will leave it all just as we have it. Make those few changes to the appveyor file, and maybe try to figure out how to add an "About" to the "preferences" that can have the necessary info. 

I am going to work on this tonight. 
Thank you so much!

canne, 2019-08-03:
Ok, but since I am on it, I'm on it... I think I find a very nice solution which satisfies your requirement to show the VERSION.cmake defined version also in plugin's preferences! You may want to consider is this what you want but it looks like that it could be.

But first, we cannot have a leading zero, in this case in the patch number. In C/C++ it is an octal number, and 010 is actually decimal 8. In VERSION.cmake define
`SET(VERSION_PATCH 10)`
The (now not needed) _GetPlugInVersionMinor()_ can be returned in its original state:```
```
int dashboard_pi::GetPlugInVersionMinor()
{
    return PLUGIN_VERSION_MINOR;
}
```
Since I noticed that **_O_** calls also the _GetCommonName()_ API-function to identify the plugin in the plugin list. Nothing prevents to add a version number in it, like this:
```
wxString dashboard_pi::GetCommonName()
{
    static const char* common_name= _("Tactics");
    char common_name_version[100];
    sprintf( common_name_version, "%s v%d.%d.%d",
            common_name, PLUGIN_VERSION_MAJOR, PLUGIN_VERSION_MINOR, PLUGIN_VERSION_PATCH ); 
    return common_name_version;
}
```
How it will look like now in the _**O**_ plug-in list (my Dashboard_Tactics as an example):
[image]
[image]
You are right, we can leave it here. I know now what I need to do in Dashboard_Tactics to get my exact version number shown in a few places for the end user (Plug-In dialog and Preferences Dialog), and get them updated automagically from the VERSION.cmake file.

Thanks for the question, I learned something new today (I did not recall that C++ 2011 has actually retained those PDP-8 era octal format with leading zero! Wow!)

rgleason, 2019-08-03:
Petri, this is starting to work!  In VERSION.cmake 
`SET(VERSION_PATCH 10)
`
In Tactics_pi.cpp   Restored the old 
```
int dashboard_pi::GetPlugInVersionMinor()
{
    return PLUGIN_VERSION_MINOR;
}
```
Then changed starting at line 765
```
//wxString tactics_pi::GetCommonName()
//{
//	return _("Tactics");
//}
wxString tactics_pi::GetCommonName()
{
    static const char* common_name= _("Tactics");
    char common_name_version[100];
    sprintf( common_name_version, "%s v%d.%d.%d",
            common_name, PLUGIN_VERSION_MAJOR, PLUGIN_VERSION_MINOR, PLUGIN_VERSION_PATCH ); 
    return common_name_version;
}
```
What I got was

[image]

canne, 2019-08-03:
Yes, I like this solution better. It took me 30 minutes to get to this type of view which I like, it is clear enough:
[image]
I think I will stick to this solution, I think it is elegant enough while waiting (forever, I am afraid) that **_O_** gods will illuminate the poor plugin coder's path with their incredible wisdom! :grin:. What do you think (for Dashboard_Tactics)?
I checked the code of Thomas in tactics_pi.cpp and since he is not referring to this function...
```
Petri@OCARINA MINGW64 ~/opencpn/tactics_pi/src (master)
$ grep CommonName *.cpp
tactics_pi.cpp:wxString tactics_pi::GetCommonName()
_<nada>_
```
...I reckon it is a good solution for you for tactics_pi maintenance until the Masters of **_O_** curse will hit you... What you can do is to change the version number for test, as v1.2.3 and build, install and then check again that there is no nasty surprise that **_O_** is somehow using the entire string to register the base product, which _**may**_ lead to problems.

I think we have something easy to maintain here: both the package name and GUI version number match all the way to the patch level, and that was the goal, right?

canne, 2019-08-03:
I checked also the ov50 code, these all the only references to this callback function:
```
Petri@OCARINA MINGW64 ~/opencpn/OpenCPN_5.0.0/OpenCPN/src (master)
$ grep GetCommonName *.cpp
pluginmanager.cpp:
pic->m_common_name = pic->m_pplugin->GetCommonName();
pluginmanager.cpp:
PluginBlacklist[i].name.c_str(), plugin->GetCommonName().c_str(), major, minor), _("Blacklisted plugin detected...");
pluginmanager.cpp:
PluginBlacklist[i].name.c_str(), plugin->GetCommonName().c_str(), major, minor);
pluginmanager.cpp:
PluginBlacklist[i].name.c_str(), plugin->GetCommonName().c_str(), major, minor), _("Blacklisted plugin detected...");
pluginmanager.cpp:
PluginBlacklist[i].name.c_str(), plugin->GetCommonName().c_str(), major, minor);
pluginmanager.cpp:wxString opencpn_plugin::GetCommonName()
```
So the only annoyance could be with the blacklisted plugin. Let's imagine that Tactics v1.0.010 gets blacklisted. Now, when you make a version v1.0.01, that will not be blacklisted because its GetCommonName() changes.

I don't think that this is a real issue. I can't even imagine how this can work? If the plugin is busted and one goes and ask from it "can you please identify yourself since I think your code is rotten and I am going to disable you!" Come on 🤦‍♂ .

Up to you, I think that there is no more need for tests to do if you like this solution.

rgleason, 2019-08-04:
Yes, you've been really on it!  And this is a huge improvement. By the way, Bdbcat said he would make some changes to the api for patch etc.  So that's good enough for me, it's ok to show major, minor, and patch.  The Developer Manual was started by Pavel, Mike and I. We have notes about compiling and the plugins, api, etc but it is not very professional as you are familiar with.

Blacklist. This is Pavel. Certain plugins were busted and users kept using them and crashing OpenCPN, saying OpenCPN crashed randomly, and it was a challenge and required iterations to get to the bottom of it. Pavel decided to really fix it! And it worked.  I suppose it could be better and of course it could be worse, but it is funny.

I am going to check out your repos latest commits to see what you've done to get there. Looks good!  It all gets handled from one file, VERSION.cmake, very nice.

rgleason, 2019-08-04:
I checked your github repository and the recent commits. I didn't see any changes from today, but I did see these
PluginPackage.cmake  about line 174
 
```
 ADD_CUSTOM_COMMAND(
   OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/${VERBOSE_NAME}-Plugin.pkg
   OUTPUT ${CMAKE_CURRENT_BINARY_DIR}/${VERBOSE_NAME}-Plugin-${OCPN_MIN_VERSION}_${VERSION_MAJOR}.${VERSION_MINOR}.${VERSION_PATCH}.pkg
   COMMAND /usr/local/bin/packagesbuild -F ${CMAKE_CURRENT_BINARY_DIR} ${CMAKE_CURRENT_BINARY_DIR}/${VERBOSE_NAME}.pkgproj
   WORKING_DIRECTORY ${CMAKE_CURRENT_BINARY_DIR}
   DEPENDS ${PACKAGE_NAME}
   COMMENT "create-pkg [${PACKAGE_NAME}]: Generating pkg file."
)

 ADD_CUSTOM_TARGET(create-pkg COMMENT "create-pkg: Done."
 DEPENDS ${CMAKE_CURRENT_BINARY_DIR}/${VERBOSE_NAME}-Plugin.pkg )
 DEPENDS ${CMAKE_CURRENT_BINARY_DIR}/${VERBOSE_NAME}-Plugin-${OCPN_MIN_VERSION}_${VERSION_MAJOR}.${VERSION_MINOR}.${VERSION_PATCH}.pkg )
```
Which is giving the MacOS packages a sensible name. I notice that this must be from Jon Gough because it uses OCPN_MIN_VERSION which I believe is used by Jon. Shouldn'y I be making these changes?

I'll wait to make the other changes.

rgleason, 2019-08-04:
Petri,
How did you do this in Preferences?
https://user-images.githubusercontent.com/6381128/62415091-b566c580-b624-11e9-9876-9194a662c04c.png

canne, 2019-08-04:
Hi Rick,

**Blacklists**: I understand. So you risk nothing. But maybe you want to test (not for blacklist) still by changing, say the minor number - test - change back 0 - test (start and stop), then search from the opencpn.log if there are any occurrences of the entire string with the version name. If not, then it is OK. If there is, we need to think again. We do not want O to think that this is a different plugin because of this common name which changes. I do not think so but better be sure.

**PluginPackage.cmake**: Yes, please make the change I stole from Jon after your recent e-mail, I was wondering why Sean's more recent module was not having it. So I took only this line from Jon, replaced the Sean's one, tested and it worked like a charm for osx in Travis.

**Showing version number in Preferences**:  I just [committed all my changes for this](https://github.com/canne/dashboard_tactics_pi/commit/40cf77ef7e5e84d7f61a202518018c5423b0d3e1). You can find my comments and explanations, incorporated. If you need further explanations, why not discuss them over there, next to the code.

leamas, 2019-08-19:
hm... first:
> This statement I cannot understand. CMake is a program which reads in Makefiles. 
> In which makefile in the plugin project is the "elsewhere" where these variables are defined?

Nope. cmake *creates* Makefiles from info in CMakelists.txt, user input and configuration checks (similar to autoconf/automake/configure).

More important: Encoding the version in ```GetCommonName()``` will create all sorts of problems. The basic assumption is that this does indeed return the unique name which does not depend on version.

I have had similar problems in https://github.com/OpenCPN/OpenCPN/pull/1390. Been able to work around by storing the "complete" version in a text file which is installed with the plugin. Now, only you know about this, so only you can display it. But it should make the trick for now without messing with a really important API function.

BTW: My proposal is that a complete version is major.minor.patch.build. This is basically semantic versioning. Build is then a string, the rest integers. See https://github.com/OpenCPN/OpenCPN/issues/1443. We certainly need to make a PR for this...

canne, 2019-08-19:
Hi, @leamas  - I understand perfectly cmake, gmake and make and the role of the input / output files in those, thanks. The outcome of this exercise is illustrated just above, in https://github.com/tom-r/tactics_pi/issues/38#issuecomment-517966220 - <major>.<minor>.<patch> , _i.e._ in the preferences dialog header only, not in the `GetCommonName()` - that was just a test. The semantic versioning is as recommended by GitHub, at least, by referring to https://semver.org. Of course, **_O_** can select whatever format best suits for the project. As soon as the format has been decided an published I will implement it accordingly.

leamas, 2019-08-19:
ok, fine, i got it wrong...

rgleason, 2019-08-20:
A big thank you to you both, as it is something I could not accomplish.

closed 2022-03-25 by rgleason

leamas, 2022-03-25:
Just a sidenote in case someone finds this: The updated 117 plugin API defines more detailed version info containing the patch, pre-release and buildinfo parts as GetPlugInVersionPatch(), GetPlugInVersionPre() and GetPlugInVersionBuild().

rgleason, 2022-03-25:
https://opencpn-manuals.github.io/main/plugin-installer/Catalog.html