tom-r/tactics_pi #41: Cannot open include file: 'vcruntime.h':

rgleason, 2019-09-15:
I am trying to set up a local Visual Studio 15 2017 dev enviroment. I was using tactics_pi to work through some of the setup bugs.
After  cmake --build . --config release   I get many of these with instruments.
```
 TacticsPreferencesDialogDef.cpp
C:\Program Files (x86)\Windows Kits\10\Include\10.0.10240.0\ucrt\corecrt.h(10): fatal error C1083: Cannot open include file: 'vcruntime.h': No such file or directory [C:\Users\Rick\Documents\GitHub\tactics_pi\build\tactics_pi.vcxproj]

```
I found that file and have added to my environment
INCLUDE   C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\include

However that does resolve it.  Do you have any ideas?

rgleason, 2019-09-15:
Well, I was wrong, a search for "corecrt.h"  turned up
`C:\Program Files (x86)\Windows Kits\10\Include\10.0.18362.0\ucrt\corecrt.h`

So  what is happening?  corecrt.h (line 10) can't find  vcruntime.h?

How to fix it?

rgleason, 2019-09-15:
Adding  to the Environment

INCLUDE C:\Program Files (x86)\Windows Kits\10\Include\10.0.18362.0\ucrt

Seems to have fixed it.

closed 2019-09-15 by rgleason

tom-r, 2019-09-16:
Rick,
Strange ...
I neither had nor have this problem, must be related to your VC2015 installation and not the plugin ...
That include 
C:\Program Files (x86)\Windows Kits\10\Include\**10.0.18362.0**\ucrt
Are you sure this is part of VC2015, or is it a leftover from an old VC installation ?
I don't have this path on my machine.
C:\Program Files (x86)\Windows Kits\10\Include    
has 3 subdirectories:
10.0.10240.0
10.0.16299.0
10.0.17763.0
...which are obviously versions of something ...
The first 2 are dated May 5th 2018 the latter one May 5th 2019 ...

Thomas

rgleason, 2019-09-17:
It appears I had an out of date version of opencpn.lib  (perhaps and early one for v4.99?) anyway this one is 146kb and it works.   I think I can remove the two includes now.