tom-r/tactics_pi #42: Barometer dial not showing pressure.

trudK45, 2019-09-18:
I have tested the Barometer instrument an it seem to work with the digital display and the histogram but not with the analog dial meter. I am sending/receiving string:
$IIXDR,P,1.01750,B,Barometer*25<0x0D><0x0A>
[image]

trudK45, 2019-09-18:
Follow up: Dial works in dashboard.

canne, 2019-09-18:
Hi, it is because of this setting in https://github.com/tom-r/tactics_pi/blob/master/src/dial.cpp starting from line 97:
```
  // Test value must be greater than 360 to enable some compass-type displays.
  if ((st == m_MainValueCap) && (data < 400.0))
...
  else if ((st == m_ExtraValueCap) && (data < 400.0))
...
```
set it back to 1200.0 which is the limit in Dashboard plug-in.

rgleason, 2022-03-25:
Changed line 104  and added a note line.

  // Set line 104 back to 1200 to display barometer too
  if ((st == m_MainValueCap) && (data < 400.0))
  {
    m_MainValue = data;
    m_MainValueUnit = unit;
  }
  else if ((st == m_ExtraValueCap) && (data < 1200.0))
  {
    m_ExtraValue = data;
    m_ExtraValueUnit = unit;
  }
}

Done, should work.

rgleason, 2022-03-25:
Fixed in v1.0.28  please close @trudK45

trudK45, 2022-03-25:
Will check this when I get back to the boat in 3-4 weeks.

closed 2022-03-25 by trudK45

rgleason, 2022-03-26:
It has been corrected in version 1.0.28 which will be in PIM shortly.