tom-r/tactics_pi #43: Send also @BWC NMEA0183 sentence. Not only XTW + RMB

sjpapa2, 2019-11-07:
In the tactics app, the phrases $PNKEP were included. These are for the polar performance calculator on the NKE platform.

Alas teh XTE menu requires also teh $BWC sentence. Can anyone, who make the PNKEP sentence, help extend the sentence maker?

Or explain how to extend the sentence maker?

rgleason, 2019-11-07:
While it might be useful in tactics, why not use nmeaconverter_pi to test trial making this sentence. If you can do it, it would provide the sentence as a workaround.  I think nemaconverter is exactly what you need. See https://opencpn.org/wiki/dokuwiki/doku.php?id=opencpn:opencpn_user_manual:plugins:logs:nmea_converter&s[]=nmeaconverter

If you can provide me with some screenshots and formulas after you get it working, I will add this example to nmeaconverter.

rgleason, 2022-03-25:
He wants tactics to populate "the  XTE menu to show $BWC sentence"

This is not what Tactics does. Also the VMG calculated by Opencpn is a "motorboat" version of VMG to a mark.

Tactics cannot interfere with the XTE display. It is a separate plugin.  There are issues with this and the API does not allow it.

This should be closed now.  @tom-r   @sjpapa2   Thanks.