tom-r/tactics_pi #44: from ownship instrument

rgleason, 2019-11-09:
Does not respond properly.  Please see.
http://www.cruisersforum.com/forums/f134/tactics-plugin-166909.html#post3012155

rgleason, 2019-11-11:
Actually when "ownship is placed here" using the right click popup menu, it works fine. Closing.

rgleason, 2019-11-11:
I have re-tested this and find it isn't working.. I am now wondering what the conditions were when it was working.  I think this should be checked. Another user has brought it up and they too are using the most recent code. v1.0.13

canne, 2019-11-11:
Hi Rick - I think that I can help you here; I fixed it a long time ago when I ported Tactics in Dashboard. IMO it has never worked with ov50 in Tactics v1.0.11 or inferior (I am bit confused with '_most recent code v1.0.13_' so I refer to this repo only and my fork of it for DashT merger):

Note that the below is an observation only, analyzing the root cause and by no means cannot be considered as instructions for deployment by copy-paste. One needs to check also instruments.h for the declaration of the sentences types (remember the starvation of available sentence types in Tactics), etc.

Tactics from_ownship.cpp and from_ownship.h are from the v4 whatsoever which started Tactics, it looks like, note these lines:

```
class TacticsInstrument_FromOwnship : public TacticsInstrument
{
public:
    TacticsInstrument_FromOwnship(wxWindow *pparent, wxWindowID id, wxString title, int cap_flag1=0, int cap_flag2=0,int cap_flag3=OCPN_DBP_STC_LAT,int cap_flag4=OCPN_DBP_STC_LON);
```
Compare this to the ov50 Dashboard's same declaration (taken from DashT since it is readily available, so look the part after the `#else` sentence, that's Dashboard):
```
public:
    DashboardInstrument_FromOwnship(wxWindow *pparent, wxWindowID id, wxString title,
#ifdef _TACTICSPI_H_
                                    unsigned long long cap_flag1=OCPN_DBP_STC_PLA,
                                    unsigned long long cap_flag2=OCPN_DBP_STC_PLO,
                                    unsigned long long cap_flag3=OCPN_DBP_STC_LAT,
                                    unsigned long long cap_flag4=OCPN_DBP_STC_LON
#else
                                    int cap_flag1=OCPN_DBP_STC_PLA,
                                    int cap_flag2=OCPN_DBP_STC_PLO,
                                    int cap_flag3=OCPN_DBP_STC_LAT,
                                    int cap_flag4=OCPN_DBP_STC_LON
#endif // _TACTICSPI_H_
```
Now, take a look at how dashboard_pi.cpp transfers the plot location position from OpenCPN to the instruments (again, ignore what is betweeen `#ifdef` and `#endif`, set for my merging effort, after that it is pure Dashboard code):
```
void dashboard_pi::SetCursorLatLon( double lat, double lon )
{
#ifdef _TACTICSPI_H_
    this->TacticsSetCursorLatLon(lat, lon);
#endif // _TACTICSPI_H_
    SendSentenceToAllInstruments( OCPN_DBP_STC_PLA, lat, _T("SDMM") );
    SendSentenceToAllInstruments( OCPN_DBP_STC_PLO, lon, _T("SDMM") );

}
````
You can observe that it uses sentences _STC_PLA and _STC_PLO of types to pass the cursor position to the instruments interested in it.

Let's look how tactics_pi.cpp in v1.0.11 is dealing with the same call-back:

```
void tactics_pi::SetCursorLatLon(double lat, double lon)
{
	g_dcur_lat = lat; //TR
	g_dcur_lon = lon;

	//SendSentenceToAllInstruments( OCPN_DBP_STC_PLA, lat, _T("SDMM") );
	//SendSentenceToAllInstruments( OCPN_DBP_STC_PLO, lon, _T("SDMM") );
}
```
So, like me above in Dashboard-Tactics code (in the if-conditional compilation), Thomas is feeding the Tactics functions with the call-back provided position about the cursor but the instruments waiting STC_PLA/PLO are starving. This makes sense for Tactics functions, of course.

Please  remember, Rick, that if one just uncomments the above sentences, it will not start to work. One need to introduce those sentences also in .h constructor's default values and they must appear also in instrument.h. You may want to see my _DashT_ repository how I've done (both .h and .cpp) if you take this endeavour. (I'm going back to _Tikhonov regularisation of ill-formed inverse problems in signal smoothing_; coding now only with Matlab/Octave and JavaScript :grin: ).

The root cause being the instrument.h starvation of usable data sentence types in favor of more usable ones (for the intended usage, Tactics), my recommendation is to, in short term advise to ignore this instrument in Tactics and - if it absolutely needed - take it from Dashboard. In medium term, maybe it can be simply removed from Tactics. I do not see it very useful in regatta situations since the numbers numbers speak less than the other, visual functions Tactics provides, allowing one to set the heading.

rgleason, 2019-11-18:
Petri, I tend to agree with you. I think this instrument should be commented out for the time being, even though it would then be code that is different from the underlying Dashboard.