tom-r/tactics_pi #49: Codeacy Issues:

rgleason, 2020-02-01:
the worst is wind.cpp

See https://app.codacy.com/manual/rgleason/tactics_pi/files?bid=13964717&step=40

canne, 2020-02-01:
An interesting conclusion. A kind reminder, default settings of Codacy, or any other Lint using tool are not necessarily applicable to a project which has not followed the default settings rules from the beginning - it is often just not worth of the effort to refactor the code to make the Lint-checker happy.

In you "particularly bad" code the issues reported are about duplicate code warnings. It is (looks to me) Codacy's own check which checks over the entire project. The results, in a C++ project not thinking about this check on can guess. See this example in wind.cpp report
[image]
Well, sure - but would one really want to go those classes and fake the results of some past cut-and-paste code creation? Not much of use. The lambda developer would do just that to get the QA guy out of his back but it has no real benefit, of course.

In my 'dev200' branch you can find this same file with a bit better results:
[image]
Why? I did not touch much this module - it is because I had to refactor the other modules and that make the duplicate check a bit better.

Do not spend too much time on these, a general **A** badge is not bad! Rather use Codacy to see that you are not going **backwards** in the coming commits, that's the best use of it.

Also, it does not hurt to pass the same code in another on-line checker, like Codecov or something else. You would notice quite different results! One needs to stick to one checker, set up his/her goals and then use the tool to meet those goals. Not that the tool is telling you what your quality goals should be, they have some common wisdom but not your wisdom.

BTW, you can control the Codacy's duplication check if you think that it shows you a rain forest with a .yml file like this (be careful with intends): https://github.com/canne/dashboard_tactics_pi/blob/7a19bccde404495c13905c6f7e3126c8514848ff/.codacy.yml

rgleason, 2020-02-01:
Thanks Petri for giving me the straight scoop.

closed 2020-02-01 by rgleason